cbfinn/gps #2: mjc_example fails on second iteration

wmontgomery4, 2016-03-04:
When running the peg insertion mjc_example, the trajectory optimizer fails on the second step because it cannot regularize with very large eta:

`ValueError: Failed to find PD solution even for very large eta (check that dynamics and cost are reasonably well conditioned)!`

The issue seems to be condition 3, which actually increases cost in the first update:
[image]

cbfinn, 2016-03-06:
I just pushed a change to the master branch that should fix this issue.

wmontgomery4, 2016-03-06:
Thanks! It's working now.

closed 2016-03-06 by wmontgomery4

DemoRunner, 2016-03-22:
@wmontgomery4  how can you set up the MuJoCo? I can not down it .can you give me some help?

wmontgomery4, 2016-03-22:
@DemoRunner I'm working on Linux, but Mujoco is proprietary software that you need a license to use.