cbfinn/gps #24: Bug

Tveek, 2016-05-11:
Running this code `python python/gps/gps_main.py box2d_badmm_example`;In the **Calculating step**,there has error below:
`/home/prafly/desktop/gps/python/gps/utility/gmm.py:202: RuntimeWarning: invalid value encountered in less
  w[:, (self.mass < (1.0 / K) * 1e-4)[:, 0]] = 1.0 / N
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "python/gps/gps_main.py", line 369, in <lambda>
    target=lambda: gps.run(itr_load=resume_training_itr)
  File "python/gps/gps_main.py", line 67, in run
    self._take_iteration(itr, traj_sample_lists)
  File "python/gps/gps_main.py", line 195, in _take_iteration
    self.algorithm.iteration(sample_lists)
  File "/home/prafly/desktop/gps/python/gps/algorithm/algorithm_badmm.py", line 59, in iteration
    self._update_policy_fit(m)  # Update policy priors.
  File "/home/prafly/desktop/gps/python/gps/algorithm/algorithm_badmm.py", line 180, in _update_policy_fit
    SampleList(self.cur[m].pol_info.policy_samples)
  File "/home/prafly/desktop/gps/python/gps/algorithm/policy/policy_prior_gmm.py", line 83, in update
    self.gmm.update(XU, K)
  File "/home/prafly/desktop/gps/python/gps/utility/gmm.py", line 174, in update
    logobs = self.estep(data)
  File "/home/prafly/desktop/gps/python/gps/utility/gmm.py", line 75, in estep
    check_finite=False)
  File "/usr/lib/python2.7/dist-packages/scipy/linalg/decomp_cholesky.py", line 81, in cholesky
    check_finite=check_finite)
  File "/usr/lib/python2.7/dist-packages/scipy/linalg/decomp_cholesky.py", line 30, in _cholesky
    raise LinAlgError("%d-th leading minor not positive definite" % info)
LinAlgError: 1-th leading minor not positive definite`

cbfinn, 2016-05-11:
Can you provide more details? -- Is the error deterministic? What iteration did it occur at?

Tveek, 2016-05-11:
@cbfinn It is  deterministic.It occurs at first iteration.

cbfinn, 2016-05-12:
Ok. Unfortunately, I don't have time to look into this right now because of
several upcoming deadlines. The example has worked for me in the past, so
it seems likely that this error is due to a recent pull request.

I'll be able to look into it after May 25 (unless the issue hasn't been
resolved by you or someone else by then).

@cbfinn <https://github.com/cbfinn> It is deterministic.It occurs at first
iteration.

tianyang9310, 2016-06-06:
I have come across this error either, is there any solution?

cbfinn, 2016-06-14:
Fixed in this PR: https://github.com/cbfinn/gps/pull/30

closed 2016-06-14 by cbfinn