cbfinn/gps #28: Compiling in Mac with Mujoco

mjm522, 2016-06-11:
I was successfully able to run ./compile_proto.sh, It did created all necessary directories.

After to install the integration with MuJoCo, I tried the following and gets following log for cmake and make -j.

```
$ cmake ../src/3rdparty
-- The C compiler identification is AppleClang 7.3.0.7030031
-- The CXX compiler identification is AppleClang 7.3.0.7030031
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Setting build type to 'Release' as none was specified.
-- Boost version: 1.60.0
-- Found the following Boost libraries:
--   python
-- Found PythonLibs: /usr/lib/libpython2.7.dylib (found suitable version "2.7.10", minimum required is "2.7") 
CMake Error at /usr/local/Cellar/cmake/3.5.2/share/cmake/Modules/FindPythonLibs.cmake:64 (get_filename_component):
  get_filename_component called with incorrect number of arguments
Call Stack (most recent call first):
  Boost.NumPy/CMakeLists.txt:14 (find_package)

-- Found PythonInterp: python (found version "2.7.11") 
-- Found NumPy: version "1.11.0" /usr/local/lib/python2.7/site-packages/numpy/core/include
-- Boost version: 1.60.0
-- Found the following Boost libraries:
--   python
-- found boost:
INCLUDE: /usr/local/include
LIB: /usr/local/lib/libboost_python-mt.dylib
osg includes: 
-- Configuring incomplete, errors occurred!
See also "/Users/michaelmathew/gps/build/CMakeFiles/CMakeOutput.log".
Michaels-MacBook-Pro:build michaelmathew$ make -j
make: *** No targets specified and no makefile found.  Stop.
```
Is there anything else that needs to done while installing on mac? I am using OSX El Capitan

cbfinn, 2016-06-11:
I would recommend trying to upgrade your cmake version. (e.g. found this on
a quick google search: https://github.com/openMVG/openMVG/issues/348)

Chelesa

mjm522, 2016-06-11:
I am already using Cmake 3.5.2..

cbfinn, 2016-06-13:
@zhangmarvin , @svlevine Do you know who has experience compiling mujoco on mac?

svlevine, 2016-06-13:
Compiling MuJoCo, or compiling mjcpy? MuJoCo is closed source, so we do not compile it. I believe there are mac binaries (you can check on the Roboti website), and then we have to figure out how to compile mjcpy for Mac, which I think shouldn't be too hard.

The error above is

> CMake Error at /usr/local/Cellar/cmake/3.5.2/share/cmake/Modules/FindPythonLibs.cmake:64 (get_filename_component):
  get_filename_component called with incorrect number of arguments

It doesn't look like this has anything to do with us. Probably a mismatch between the cmake version and the FindPythonLibs.cmake module.

mjm522, 2016-07-09:
I solved this issue by downgrading to Cmake 3.3.2 and then rebuilding.
However now I have a new error.
when I try to run the example
python python/gps/gps_main.py mjc_example
[It says: No module named mjcpy](url)

cbfinn, 2016-07-09:
That error usually means that your python path isn't set up properly. Make sure that mjcpy is on your python path and ld library path, as explained in the set-up instructions.

Also, .bashrc files aren't always executed on start-up for macs, so you may want to source it in your .bash_profile.

mjm522, 2016-07-09:
I have edited the `.bash_profile.`
I have doubt:

The setup says to add

```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/path/to/gps/build/lib
export PYTHONPATH=$PYTHONPATH:/path/to/gps/build/lib
```

The `make -j `has produced only one file in the above path. `libboost_numpy.dylib `
Shouldn't some mujoco library should also be compiled and some libraries should be present?

But I didn't get any error while running the cmake or make.

How many files do the `make -j` generate?

mjm522, 2016-07-10:
I saw there were [some changes ](https://github.com/cbfinn/gps/pull/25)for the files and hence resynced the files. I am also using mujoco131 Now I have the following problem:

/Users/michaelmathew/gps/src/3rdparty/mjcpy2/../mjpro131/include/mjmodel.h:13:9: warning: 'mjUSEDOUBLE' macro redefined
      [-Wmacro-redefined]
#define mjUSEDOUBLE
        ^
<command line>:1:9: note: previous definition is here
#define mjUSEDOUBLE 1
        ^
/Users/michaelmathew/gps/src/3rdparty/mjcpy2/mujoco_osg_viewer.cpp:5:10: fatal error: 'mujoco.h' file not found
#include "mujoco.h"
         ^
1 warning and 1 error generated.
make[2]: *** [mjcpy2/CMakeFiles/mjcpy.dir/mujoco_osg_viewer.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /Users/michaelmathew/gps/src/3rdparty/mjcpy2/mjcpy2.cpp:6:
In file included from /Users/michaelmathew/gps/src/3rdparty/mjcpy2/mujoco_osg_viewer.hpp:5:
In file included from /Users/michaelmathew/gps/src/3rdparty/mjcpy2/../mjpro131/include/mujoco.h:38:
/Users/michaelmathew/gps/src/3rdparty/mjcpy2/../mjpro131/include/mjmodel.h:13:9: warning: 'mjUSEDOUBLE' macro redefined
      [-Wmacro-redefined]
#define mjUSEDOUBLE
        ^
<command line>:1:9: note: previous definition is here
#define mjUSEDOUBLE 1
        ^
In file included from /Users/michaelmathew/gps/src/3rdparty/mjcpy2/mjcpy2.cpp:279:
/Users/michaelmathew/gps/src/3rdparty/mjcpy2/mjcpy2_getmodel_autogen.i:159:54: error: no member named 'mat_texscale' in
      '_mjModel'
    out["mat_texscale"] = toNdarray2<float>(m_model->mat_texscale, m_model->nmat, 2);
                                            ~~~~~~~  ^
In file included from /Users/michaelmathew/gps/src/3rdparty/mjcpy2/mjcpy2.cpp:283:
/Users/michaelmathew/gps/src/3rdparty/mjcpy2/mjcpy2_setmodel_autogen.i:159:41: error: no member named 'mat_texscale' in
      '_mjModel'
    _cadihk(d, "mat_texscale", m_model->mat_texscale);
                               ~~~~~~~  ^
In file included from /Users/michaelmathew/gps/src/3rdparty/mjcpy2/mjcpy2.cpp:287:
/Users/michaelmathew/gps/src/3rdparty/mjcpy2/mjcpy2_getdata_autogen.i:61:52: error: no member named 'efc_signature' in
      '_mjData'
    out["efc_signature"] = toNdarray2<int>(m_data->efc_signature, m_model->njmax, 1);
                                           ~~~~~~  ^
In file included from /Users/michaelmathew/gps/src/3rdparty/mjcpy2/mjcpy2.cpp:292:
/Users/michaelmathew/gps/src/3rdparty/mjcpy2/mjcpy2_setdata_autogen.i:61:41: error: no member named 'efc_signature' in
      '_mjData'
    _cadihk(d, "efc_signature", m_data->efc_signature);
                                ~~~~~~  ^
1 warning and 4 errors generated.
make[2]: *** [mjcpy2/CMakeFiles/mjcpy.dir/mjcpy2.cpp.o] Error 1
make[1]: *** [mjcpy2/CMakeFiles/mjcpy.dir/all] Error 2
make: *** [all] Error 2

The thread said the system is working on mujoco 1.31. Could you please tell me what were the changes that were done?

mjm522, 2016-07-11:
Could you please tell me on which version of python you got it working?

cbfinn, 2016-07-11:
We assume python 2.7, as mentioned on the docs:
http://rll.berkeley.edu/gps/

eaa3, 2016-07-12:
There appears to be a problem with line 64 in the file /usr/local/Cellar/cmake/3.5.2/share/cmake/Modules/FindPythonLibs.cmake.
I have temporarily commented this line in order to build gps on OSX Mavericks. After that, it generates fine with cmake 3.5.1. My guess is that there was a change in the behaviour of the function get_filename_component on more recent versions of cmake that made the statement in line 64 syntactically incorrect.

mjm522, 2016-07-13:
I finally compiled everything and it works fine now.

Here are a set of instructions to follow to install gps on Mac. It is tested on OSX El-Capitan and OSX Maverics :)

closed 2016-07-13 by mjm522

cbfinn, 2016-07-13:
Great, thanks!

cbfinn, 2016-07-13:
Can you make a pull request that adds a link to your instructions to the gps docs?

mjm522, 2016-07-13:
Submitted :)

cbfinn, 2016-07-14:
I don't see the pull request...

mjm522, 2016-07-14:
Could you please check now?

lakehanne, 2017-06-07:
Did you guys merge the pull request? I spent a number of hours trying to get this fixed.

jangirrishabh, 2018-02-06:
@mjm522 @cbfinn I found an easier solution, just comment out Line number 14 in gps/src/3rdparty/Boost.NumPy/CMakeLists.txt
Where it is trying to look for PythonLibs 2.7 REQUIRED, because the requirement is already satisfied and thus conflicting with 3rdparty/CMakeLists.txt

I am in a Anaconda virtual environment on a Linux GPU machine, so my guess is that there was a conflict between the root python libs and the conda python libs

[image]