cbfinn/gps #34: ImportError: No module named Box2D (I had installed pybox2d)

dongleecsu, 2016-06-25:
Hi Chelsea,
Thank you for your awesome work about GPS. I followed the instructions to install the GPS algorithm from [http://rll.berkeley.edu/gps/](url), but got an **import error: No module named Box2D** when I ran the code
`python python/gps/gps_main.py box2d_pointmass_example`

This is how I set up Pybox2D( the 2nd step is different, since the url:http://pybox2d.googlecode.com/svn/trunk/ was not found):
1. `sudo apt-get install build-essential python-dev swig python-pygame subversion`
2. `git clone https://github.com/pybox2d/pybox2d`
3. `python setup.py build
sudo python setup.py install`
So I thought maybe I had installed pybox2d correctly. But there was always an error :ImportError: No module named Box2D".

Could you please give me any advice about this issue? Thank you.

Dong Li

cbfinn, 2016-06-25:
It sounds like your installation wasn't successful. You should be able to successfully run "import Box2D" in a python shell, once it's installed.

Your steps look reasonable based on the pybox2d github installation instructions.

cbfinn, 2016-06-25:
Can you try running the following to install?
```
sudo apt-get install python-box2d
```

dongleecsu, 2016-06-26:
I tried to install by following: `sudo apt-get install python-box2d`, and everything went well.
But when I ran `import Box2D`, the same error threw.
I also deleted pybox2d and reinstall again, but got the same error.
It's really weird to get this error since I followed the pybox2d github installation instructions.

cbfinn, 2016-06-26:
I would bring up this error with the Box2D codebase. Sorry, I don't know how else to help.

mjm522, 2016-07-09:
The instruction in the Gps didn't work for me as well. But this is what I did. The box2d example works for me.

For installing box2d

First install swig,
then:
$git clone https://github.com/pybox2d/pybox2d pybox2d_dev
$cd pybox2d_dev
$python setup.py build 
$sudo python setup.py install
For installing pygame

try: $brew install homebrew/python/pygame
if it doesn't work completely try out the following

Else install qt, pyside, mercurial, sdl
Then,
$hg clone https://bitbucket.org/pygame/pygame
$cd pygame
$python setup.py build
$python setup.py install
Now try to run one of their examples.

$cd gps
$python python/gps/gps_main.py box2d_pointmass_example

Hope it helps.

cbfinn, 2016-07-09:
Looks like you're using a Mac. The set-up instructions are meant for linux. I'm glad you got it working for you though.

mjm522, 2016-07-09:
Yeah this is for mac. :)

dongleecsu, 2016-07-13:
@mjm522  Thank you for your suggestions and it helps.
@cbfinn  I finally figured this problem out. This is exact what I did:

**First, install pybox2d.**
Follow @mjm522 's step, I install `pybox2d_dev` (I have installed `swig` before):
```shell
$git clone https://github.com/pybox2d/pybox2d pybox2d_dev
$cd pybox2d_dev
$python setup.py build 
$sudo python setup.py install
```
At this time, I could import Box2D successfully.

**Second, install pygame**
Follow pybox2d's github instruction:
`conda install -c https://conda.anaconda.org/kne pygame`

For some unknown reasons, I must add `sudo` in order to go through the examples without error. 
Try some examples:
```shell
$cd gps
$sudo python python/gps/gps_main.py box2d_pointmass_example 
```

cbfinn, 2016-07-13:
Ok, I'm glad you figured it out. I'm closing this issue now.

closed 2016-07-13 by cbfinn

Avinash72, 2018-04-17:
Hi @dongleecsu I tried the steps that you've mentioned but I'm not able to get the Box2D up. I still get the error of 

  from ._conv import register_converters as _register_converters
Using TensorFlow backend.
Traceback (most recent call last):
  File "lunar_lander_dl_player.py", line 13, in <module>
    import Box2D
ModuleNotFoundError: No module named 'Box2D'

EldhosePoulose, 2022-08-05:
!pip3 install gym[box2d]