cbfinn/gps #47: Problem in using MuJoCo131 trial license

happyzgp, 2016-08-20:
Hi, I try to use MuJoCo131 trial license with GPS, but on running the mjc_example experiment, it will output:

ERROR: Invalid activation key

The trial license works well with MuJoCo's own simulate program. 

Does anyone know the reason? Thanks!

cbfinn, 2016-08-21:
You may want to make sure that the key is in the correct directory. Otherwise, I don't know what would cause the issue.

happyzgp, 2016-08-22:
Hi Chelsea, thanks for your comments. I put the key in the path 'gps/src/3rdparty/mjpro/mjkey.txt', is that OK? Where should I put the key? or is it because I use the the trial version? Thanks!

cbfinn, 2016-08-23:
That should be fine. I haven't tried using the trial version before, so I don't know if it would cause issues.

cbfinn, 2016-08-23:
I would recommend contacting mujoco developers: http://www.mujoco.org/contact.html

happyzgp, 2016-08-30:
Hi Chelsea, thank you. I will check it out.

closed 2016-09-27 by happyzgp

acrosson, 2017-06-16:
@happyzgp how did you resolve?

flyingdust777, 2017-06-22:
@happyzgp were you able to resolve this?

newuhe, 2018-08-20:
I encounter the problem`Could not open activation key file src/3rdparty/mjpro/mjkey.txt`