cbfinn/gps #55: Doubt on 'conditions' parameter

mjm522, 2016-10-12:
I am unable to find out what exactly is the role of **'conditions' :4** in the hyperparameters file? Could someone please help?

wmontgomery4, 2016-10-12:
It's maybe poorly named, but `conditions` just refers to the number of experimental conditions, which is typically just the number of initial positions.

mjm522, 2016-10-12:
Thank you for the reply. Could you please elaborate a little more?
Number of initial positions? Do you mean four different positions (of the robot) from which the same experiment is done parallelly? Or four different start locations of end effector (for e.g. the peg insertion task)?  Why is it done like that? Is is just for the having different views of the world if you are using cov-nets for image processing?

wmontgomery4, 2016-10-12:
It can refer to different positions of the robot, or different target positions, it depends on how you set up the rest of the hyperparams. It's done like this because that's the point of GPS: we essentially solve the problem locally from M different initial conditions, then use a neural network to combine those local solutions into one global solution.

cbfinn, 2016-10-12:
Please refer to the paper for basic algorithmic questions:
http://jmlr.org/papers/v17/15-522.html

closed 2016-10-12 by cbfinn