cbfinn/gps #61: Docker Image?

lakehanne, 2016-10-23:
Hey,

Thanks for making this code public. Are there plans to release a docker image of this setup anytime soon?
It would be nice to have, I think.

cbfinn, 2016-10-24:
We don't currently have plans to make a docker image of the set-up in the near future, but we would welcome such a contribution!

We chose not to use catkin because there were some people still using ROS fuerte in our lab. I agree that it would be nice.

lakehanne, 2016-10-31:
Thought I should leave the link to the catkinized version of the ROS package here as others might find it useful. https://github.com/lakehanne/gps_catkin

cbfinn, 2016-10-31:
I'll leave this issue open for now, until there is a docker image.

ahundt, 2017-01-09:
I've created the following pull request https://github.com/cbfinn/gps/pull/69 which likely does much of the same. 

@lakehanne I looked at your repository and your work seems to be a separate repository rather than modifying a fork, was that intentional?

lakehanne, 2017-01-10:
Not really. I guess I could have forked chelsea's work and send a PR of my work but this was supposed to be a wip till I finish the docker image. Sadly, I have been busy with upcoming deadlines that I have not paid too much attention to developing the docker image lately. I should be able to get back to it soon though. 

Hey, as per #69 I am not too familiar with Karmic. Have you tested your code with jade and indigo yet? I do not have the time to test now. If you are just hopeful it would work with indigo ot Jade, you might as well just call it a `karmic-devel branch`. 

@cbfinn , would you rather have me send my [indigo repo](https://github.com/lakehanne/gps/tree/master) as a PR to you in the meantime?

lakehanne, 2017-05-03:
Initial docker image here:

https://hub.docker.com/u/lakehanne/

I can get the box examples to run but the matplotlib window does not update the charts on every system I have tested it on. If anyone can figure out how to do this, it would be appreciated.

The dockerfile for the build is available here:

https://github.com/lakehanne/gps/blob/master/Dockerfile

lakehanne, 2017-06-17:
I found a fix for the `Bad Drawable error` I was getting earlier. It seems we needed to export the environmental variable for QT or KDE.

Doing this fixes the drawable errors:
 
#### Fix
Add this line `QT_X11_NO_MITSHM=1` in `~/.bashrc` or export the environmental variable from terminal:

`export QT_X11_NO_MITSHM=1`

Then run the bash script for the examples. @cbfinn, I think you can close this issue now.

ahundt, 2017-06-19:
@lakehanne I think the changes would need to be merged in to this repository for closing this to be appropriate.

ruinianxu, 2017-11-07:
Hi lakehanne,
I wonder how did you install pr2 package. Since there is no official released pr2 package which contains all related needed packages, my strategy is to separately install binary pr2 packages like pr2_simulator. I tried source code but there was still some dependency problem. Since you are also using ros indigo, really appreciate it if you can give me some suggestions.

lakehanne, 2017-11-07:
The pr2 package is shipped in ubuntu 14.04 repos as part of the ROS packages.

cd to the manifest folder and do

‘rosdep install -from-paths . -ry ‘

Sincerely,
Lekan

“A graduate student’s job is to turn his advisor’s bad ideas into good ones.” ~ Pedro Domingos

huiwenzhang, 2018-04-21:
@lakehanne  Hi, thanks for your job. It's really helpful. I want to know which version of your images is the latest on docker hub? You have gps-v0.0.2 and v0.1.1 , but  it not clear about the difference.

lakehanne, 2018-04-21:
I can’t remember now but I think it must be the one with the highest tag.

-----
Sincerely,
Lékan

"The past was erased, the erasure was forgotten, the lie became the truth." 

--George Orwell

huiwenzhang, 2018-04-21:
@lakehanne  Ok, I will try it.

closed 2019-08-23 by lakehanne