cbfinn/gps #66: How were the PR2 gains found for the PR2 example?

mjm522, 2016-11-07:
Hi,

We wrote a Baxter ros package for gps for testing the same code in baxter for a reaching task (torque control mode). Though the cost converges, it doesn't very well reach the goal location. We doubt the gain values. So it would be very helpful to know how you identified the gains of the pr2 given in hyperparams file of the pr2_example. 

Another issue is that giving a slightly far of location on the pr2_example does not very well behave correctly. The example uses a very close target location to its starting location. Is there a rule of thumb on how far the target location should be from the start location?  I doubt this is because of lack of data if we give a far of location.

Any helpful suggestion is appreciated.

Best,

Mike

closed 2016-11-10 by mjm522

cbfinn, 2016-11-20:
Sorry for the slow response. We got the PR2_GAINS from the feedforward gains set in the Willow Garrage feedfoward Jacobian transpose controller. They're a diagonal approximation to the inverse mass matrix of the PR2 arm. The joints are ordered from shoulder to wrist.

arpit15, 2016-12-26:
@mjm522 were you able to find the correct GAINS for baxter? I want to use the code for baxter robot. I obtained the following jacobian matrix using baxter_pykdl lib for right arm
[[ -9.81223815e-02,   5.27690945e-01,   2.76055334e-03,
         -1.73863717e-02,  -8.46334285e-03,   9.99813025e-01],
       [ -2.52418116e-01,   1.35423507e-01,  -3.59344505e-01,
          4.55531672e-01,   8.90085393e-01,   1.54560307e-02],
       [ -5.17605060e-02,   5.31847136e-01,   1.62781266e-01,
          4.16955267e-01,  -2.28666703e-01,   8.79693040e-01],
       [ -5.71443074e-01,  -9.60165082e-02,  -2.86427784e-01,
         -3.36570105e-01,   8.60195080e-01,   3.83125290e-01],
       [ -1.99182311e-01,   2.82692540e-01,  -4.58224579e-05,
          7.08741024e-01,   4.99290989e-01,  -4.98392083e-01],
       [ -3.16542974e-01,  -2.23032950e-01,   2.21933475e-04,
         -5.75977813e-01,   8.17465315e-01,  -1.32505339e-04],
       [  1.41386740e-17,   4.41321106e-17,  -2.48940005e-20,
         -3.92200559e-04,  -4.38433442e-04,  -9.99999827e-01]]
I also checked out your [example](https://github.com/mjm522/gps/blob/baxter/experiments/my_baxter_experiment/hyperparams.py). I don't understand how you obtained [1e2, 1e2, 1e2, 1e2, 1e2, 1e2, 1e2] as diagonal approximation. 
Can @mjm522  kindly mention the method used to arrive at the gains from the jacobian matrix?

@cbfinn Is [this](http://wiki.ros.org/robot_mechanism_controllers/JTCartesian%20Controller) the page you referred for setting PR2_GAINS? Kindly mention what is the full jacobian for PR2 or the method to derive approximation using jacobian as this technique might be useful for any robot.

Thank you

mjm522, 2016-12-27:
@arpit15, those gains are wrong,  (1e-2) ones, as @cbfinn has said, you could get the "diagonal approximation to the inverse mass matrix of the PR2(Baxter) arm", I accessed those values using Baxter interface itself.

arpit15, 2016-12-27:
@mjm522 Can you elaborate on the steps you used for getting the BAXTER_GAINS?