cbfinn/gps #70: I got error on mjc_peg_images

spk921, 2017-01-06:
I am using TF v0.12 and got following error when I try to run mjc_peg_images. 
Also I tried with TFv0.11 same error. 
Which version of TF works ? 

DEBUG:No gps_agent_pkg: gps_agent_pkg
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/opt/ros/kinetic/share
I tensorflow/stream_executor/dso_loader.cc:125] successfully opened CUDA library libcublas.so.8.0 locally
I tensorflow/stream_executor/dso_loader.cc:125] successfully opened CUDA library libcudnn.so.5 locally
I tensorflow/stream_executor/dso_loader.cc:125] successfully opened CUDA library libcufft.so.8.0 locally
I tensorflow/stream_executor/dso_loader.cc:125] successfully opened CUDA library libcuda.so.1 locally
I tensorflow/stream_executor/dso_loader.cc:125] successfully opened CUDA library libcurand.so.8.0 locally
W tensorflow/core/platform/cpu_feature_guard.cc:95] The TensorFlow library wasn't compiled to use AVX2 instructions, but these are available on your machine and could speed up CPU computations.
Traceback (most recent call last):
  File "python/gps/gps_main.py", line 410, in <module>
    main()
  File "python/gps/gps_main.py", line 395, in main
    gps = GPSMain(hyperparams.config, args.quit)
  File "python/gps/gps_main.py", line 51, in __init__
    self.algorithm = config['algorithm']['type'](config['algorithm'])
  File "python/gps/algorithm/algorithm_badmm.py", line 34, in __init__
    self._hyperparams['policy_opt'], self.dO, self.dU
  File "python/gps/algorithm/policy_opt/policy_opt_tf.py", line 47, in __init__
    self.init_network()
  File "python/gps/algorithm/policy_opt/policy_opt_tf.py", line 71, in init_network
    network_config=self._hyperparams['network_params'])
  File "python/gps/algorithm/policy_opt/tf_model_example.py", line 147, in multi_modal_network
    'bc2': init_bias([num_filters[1]]),
  File "python/gps/algorithm/policy_opt/tf_model_example.py", line 13, in init_bias
    return tf.get_variable(name, initializer=tf.zeros(shape, dtype='float'))
  File "/usr/local/lib/python2.7/dist-packages/tensorflow/python/ops/variable_scope.py", line 987, in get_variable
    custom_getter=custom_getter)
  File "/usr/local/lib/python2.7/dist-packages/tensorflow/python/ops/variable_scope.py", line 889, in get_variable
    custom_getter=custom_getter)
  File "/usr/local/lib/python2.7/dist-packages/tensorflow/python/ops/variable_scope.py", line 347, in get_variable
    validate_shape=validate_shape)
  File "/usr/local/lib/python2.7/dist-packages/tensorflow/python/ops/variable_scope.py", line 332, in _true_getter
    caching_device=caching_device, validate_shape=validate_shape)
  File "/usr/local/lib/python2.7/dist-packages/tensorflow/python/ops/variable_scope.py", line 638, in _get_single_variable
    name, "".join(traceback.format_list(tb))))
ValueError: Variable None already exists, disallowed. Did you mean to set reuse=True in VarScope? Originally defined at:

  File "python/gps/algorithm/policy_opt/tf_model_example.py", line 13, in init_bias
    return tf.get_variable(name, initializer=tf.zeros(shape, dtype='float'))
  File "python/gps/algorithm/policy_opt/tf_model_example.py", line 146, in multi_modal_network
    'bc1': init_bias([num_filters[0]]),
  File "python/gps/algorithm/policy_opt/policy_opt_tf.py", line 71, in init_network
    network_config=self._hyperparams['network_params'])

INFO:signal_shutdown [atexit]

cbfinn, 2017-01-06:
@bstadie Which version of TF did you use when you wrote this code?

foolyc, 2017-08-03:
add name for the init_bias
`
    biases = {
        'bc1': init_bias([num_filters[0]], name='bc1'),
        'bc2': init_bias([num_filters[1]], name='bc2'),
    }
`
but you will still faces other bugs when run the mjc_peg_images in tf0.12

fhoekstra, 2017-09-28:
I'm getting this same error on Tensorflow 0.8.0 (on Ubuntu)

@foolyc Should adding the name also be necessary for tensorflow 0.8 ? As @cbfinn says it was tested for v0.5-0.8

FYI I am modifying the box2d agent to use images as input. I do not have Mujoco installed, but got the same error when trying to use a neural network.