cbfinn/gps #76: Boost.Python.ArgumentError: Python argument types did not match C++ signature

henrykmichalewski, 2017-02-19:
Can you help with the following boost-related error? 

```
$ python python/gps/gps_main.py mjc_example
DEBUG:No ROS enabled: No module named rospkg
Traceback (most recent call last):
  File "python/gps/gps_main.py", line 410, in <module>
    main()
  File "python/gps/gps_main.py", line 395, in main
    gps = GPSMain(hyperparams.config, args.quit)
  File "python/gps/gps_main.py", line 46, in __init__
    self.agent = config['agent']['type'](config['agent'])
  File "python/gps/agent/mjc/agent_mjc.py", line 30, in __init__
    self._setup_world(hyperparams['filename'])
  File "python/gps/agent/mjc/agent_mjc.py", line 54, in _setup_world
    world = mjcpy.MJCWorld(filename)
Boost.Python.ArgumentError: Python argument types in
    MJCWorld.__init__(MJCWorld, str)
did not match C++ signature:
    __init__(_object*, std::string)
```

My machine is a generic Ubuntu 16.04 box with Boost 1.58, MuJoCo 131.

henrykmichalewski, 2017-02-26:
I have a question to people who successfully ran the original code: which versions of the OS, the Boost library and the MuJoCo engine do you use?

UntilDeath, 2017-02-27:
I did not use the MuJoCo engine, but encountered the same error while installing fresh on a new machine. I think the problem stems from the command "sudo apt-get install libboost-all-dev" which installs an older version than is available from the source code. By installing the latest boost from the source code, I've solved this error.

henrykmichalewski, 2017-03-02:
@burak-cetinkaya Thank you very much for your comment. The master branch works just fine with Ubuntu 14.04, cmake 2.8.12.2 and boost 1.54.

closed 2017-03-02 by henrykmichalewski

huangjiancong1, 2020-06-12:
> @burak-cetinkaya Thank you very much for your comment. The master branch works just fine with Ubuntu 14.04, cmake 2.8.12.2 and boost 1.54.

Can I install this under Ubuntu 16.04 ? I have some issue like:
```
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   python
CMake Error at /usr/share/cmake-3.5/Modules/FindPythonLibs.cmake:64 (get_filename_component):
  get_filename_component called with incorrect number of arguments
Call Stack (most recent call first):
  Boost.NumPy/CMakeLists.txt:14 (find_package)

-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   python
-- found boost:
INCLUDE: /usr/include
LIB: /usr/lib/x86_64-linux-gnu/libboost_python.so
osg includes: /usr/include
-- Configuring incomplete, errors occurred!
See also "/home/jim/mygithub/other_author_demo/gps/build/CMakeFiles/CMakeOutput.log".
```
Is it relate to the Boost version? Should I change the version?