cbfinn/gps #82: Support tensorflow 1.2

PhilJd, 2017-05-26:
Hi,
thanks for the great work!
I tweaked a few lines in the code in order to run gps using the newest version of tensorflow. Would you be interested in a pull request? It's mainly about changing deprecated functions.

Cheers,
Phi

cbfinn, 2017-06-01:
Yes, we would appreciate a pull request!

JianlanLuo, 2017-06-09:
@PhilJd  Hi Phil,  did you submit the PR by any chance, I am using tf1.1, and would appreciate your code.

Thanks,
Jianlan

PhilJd, 2017-06-09:
Hi @JianlanLuo ,
I just committed my current status to my fork at phil/tensorflow_1.2. This code runs, but the net does not converge, also mentioned [here](https://groups.google.com/forum/#!topic/gps-help/YRLc1AT6X9E). 
I wanted to fix that before submitting the pull request, but I was not able to spot the bug yet. Maybe I missed one of the cases where the function parameters are not explicitely specified, e.g. `tf.concat` changed from` tf.concat(axis, value)` to `tf.concat(value, axis)`.

I'll try to have a look again either on Sunday or early next week.
Cheers,
Phil

lakehanne, 2017-06-10:
I tried the badmm_example with tf and can confirm the following error:

```python
DEBUG:LinAlgError: 1-th leading minor not positive definite
DEBUG:Increasing eta: 9872104440081784.000000 -> 10333273041924524.000000
Traceback (most recent call last):
  File "python/gps/gps_main.py", line 79, in run
    self._take_iteration(itr, traj_sample_lists)
  File "python/gps/gps_main.py", line 217, in _take_iteration
    self.algorithm.iteration(sample_lists)
  File "python/gps/algorithm/algorithm_badmm.py", line 68, in iteration
    self._update_trajectories()
  File "python/gps/algorithm/algorithm.py", line 127, in _update_trajectories
    self.traj_opt.update(cond, self)
  File "python/gps/algorithm/traj_opt/traj_opt_lqr_python.py", line 78, in update
    eta, algorithm, m)
  File "python/gps/algorithm/traj_opt/traj_opt_lqr_python.py", line 453, in backward
    reasonably well conditioned)!')
ValueError: Failed to find PD solution even for very                             large eta (check that dynamics and cost are                             reasonably well conditioned)!
```

Is this similar to the error you had?

PhilJd, 2017-06-10:
No, for me there was no ValueError, the net ran fine but didn't converge. Thanks for the Traceback, there's a good chance that this is related to my issue. Did this error occur on my branch or on the original master? And which tf version are you using?

lakehanne, 2017-06-10:
A variant of the original master: lakehanne/gps . TF 1.1