cbfinn/gps #83: Errors compiling with mujoco pro 150

lakehanne, 2017-06-07:
How were the `mjcpy2_getmodel_autogen.i` files for mujoco generated? I am using version mujoco150 but I have problems compiling:

```bash
/home/lex/catkin_ws/src/gps/3rdparty/mjcpy2/mjcpy2_getmodel_autogen.i: In member function ‘boost::python::dict {anonymous}::PyMJCWorld2::GetModel()’:
/home/lex/catkin_ws/src/gps/3rdparty/mjcpy2/mjcpy2_getmodel_autogen.i:85:58: error: ‘mjModel {aka struct _mjModel}’ has no member named ‘dof_frictional’
     out["dof_frictional"] = toNdarray2<mjtByte>(m_model->dof_frictional, m_mode
                                                          ^
/home/lex/catkin_ws/src/gps/3rdparty/mjcpy2/mjcpy2_getmodel_autogen.i:185:61: error: ‘mjModel {aka struct _mjModel}’ has no member named ‘tendon_frictional’
     out["tendon_frictional"] = toNdarray2<mjtByte>(m_model->tendon_frictional, 
                                                             ^
/home/lex/catkin_ws/src/gps/3rdparty/mjcpy2/mjcpy2_getmodel_autogen.i:226:55: error: ‘mjModel {aka struct _mjModel}’ has no member named ‘sensor_scale’
     out["sensor_scale"] = toNdarray2<mjtNum>(m_model->sensor_scale, m_model->ns
                                                       ^
In file included from /home/lex/catkin_ws/src/gps/3rdparty/mjcpy2/mjcpy2.cpp:283:0:
/home/lex/catkin_ws/src/gps/3rdparty/mjcpy2/mjcpy2_setmodel_autogen.i: In member function ‘void {anonymous}::PyMJCWorld2::SetModel(boost::python::dict)’:
/home/lex/catkin_ws/src/gps/3rdparty/mjcpy2/mjcpy2_setmodel_autogen.i:85:43: error: ‘mjModel {aka struct _mjModel}’ has no member named ‘dof_frictional’
     _cadihk(d, "dof_frictional", m_model->dof_frictional);
```             

Is there a way I can autogenerate these files for my version of mujoco?

acrosson, 2017-06-14:
Hi @lakehanne same issue. Can't find older version. The autogen files need to be updated.

acrosson, 2017-06-14:
You can run the struct generator `gps/src/3rdparty/mjcpy2/mj_struct_codegen.py` and toggle between the model and data by commenting(uncommenting) last few lines. This should get rid of those compile errors. 

Now i'm getting:
```
<command-line>:0:0: note: this is the location of the previous definition
CMakeFiles/Makefile2:158: recipe for target 'mjcpy2/CMakeFiles/mjcpy.dir/all' failed
make[1]: *** [mjcpy2/CMakeFiles/mjcpy.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

acrosson, 2017-06-15:
I had two versions of mujoco installed in mjpro folder. Removed and replaced with single version. I'm able to compile file now.

lakehanne, 2017-06-25:
Can you share the code gen code?

facetohard, 2017-10-05:
I have solved this problem.
First, you should delete all files with '.i' in the dir `\path\to\gps\src\3rdparty\mjcpy2`, then open the `mj_struct_codegen.py` and uncoment
`with open("mjcpy2_getmodel_autogen.i","w") as outfile:
    get_lines, set_lines = process(find_lines_between(all_lines, "_mjModel","}"),"m_model")
    outfile.write("\n".join(get_lines))
with open("mjcpy2_setmodel_autogen.i","w") as outfile:
    get_lines, set_lines = process(find_lines_between(all_lines, "_mjModel","}"),"m_model")
    outfile.write("\n".join(set_lines))`
and modify the line 5 to `fname = "../mjpro/include/mjdata.h"`
Then, you can run `python mj_struct_codegen.py` to generate some files and follow the instruction of gps to compile gps.

closed 2019-08-23 by lakehanne

pkusjh, 2019-09-15:
> You can run the struct generator `gps/src/3rdparty/mjcpy2/mj_struct_codegen.py` and toggle between the model and data by commenting(uncommenting) last few lines. This should get rid of those compile errors.
> 
> Now i'm getting:
> 
> ```
> make[2]: *** No rule to make target '/home/user/gps/src/3rdparty/mjpro/bin/libmujoco131.so', needed by 'lib/mjcpy.so'.  Stop.
> <command-line>:0:0: note: this is the location of the previous definition
> CMakeFiles/Makefile2:158: recipe for target 'mjcpy2/CMakeFiles/mjcpy.dir/all' failed
> make[1]: *** [mjcpy2/CMakeFiles/mjcpy.dir/all] Error 2
> Makefile:127: recipe for target 'all' failed
> make: *** [all] Error 2
> ```

How do you solve this problem?

pkusjh, 2019-09-15:
> You can run the struct generator `gps/src/3rdparty/mjcpy2/mj_struct_codegen.py` and toggle between the model and data by commenting(uncommenting) last few lines. This should get rid of those compile [...]

@acrosson How did you solve this problem?

padmaja-kulkarni, 2019-09-23:
Hi! This issue could be solved by using Mujoco version 1.31 from 2016.