cbfinn/gps #84: GPS  in humanoid running exp(cost function derivativate)

foolyc, 2017-06-16:
Has anyone tried the humanoid walking or running experiments?

if the state vector is the joint angle and joint velocity, how can i get the derivativate dp/dq(where p  is the height above the ground of the pelvis)?

Other than the robot arm, the end position is the map function of the joint angle f(q), as for humanoid walking or running
- the pelvis is not the function of the joint angle(when the support foot is not full contact with the ground or the robot is just falling down) 
- the joint angle of legs is not freedom in double support phase(when two feet are full contacted).

how to deal with this problem?

cbfinn, 2017-06-17:
I don't know anyone who has tried humanoid locomotion with GPS. It is not clear how well LQR will work in that setting.

foolyc, 2017-06-18:
@cbfinn  Thanks for your time.
In the phd thesis(and some other paper) of Dr. Sergey Levine, GPS exp on humanoid walking and running is included, but with little details.
So i come up with this problems when duplicating the humanoid exp.

cbfinn, 2017-06-19:
@svlevine

dujinyu, 2020-01-07:
@foolyc have you successfully implemented the exp on humanoid walking based on cbfinn's gps code.