cbfinn/gps #86: mjcpy.so returns error when run "python python/gps/gps_main.py mjc_example"

barian-do, 2017-06-28:
Hi, 
I need a help to run the mujoco example for your GPS, 
I can run box2d related examples, but stuck at the mujoco ones.

the error is:

> ImportError: /home/baek/gps/build/lib/mjcpy.so: undefined symbol: _ZN3osg8Geometry16setTexCoordArrayEjPNS_5ArrayE

I followed exactly what's said at the https://github.com/cbfinn/gps site.
it looks like the OSG_LIBRARIES is somehow wrong.
In two of the OSG_LIBRARIES, 

> /usr/lib/x86_64-linux-gnu/libosg.so;
> /usr/lib/x86_64-linux-gnu/libosgViewer.so;

I found 

> _ZN3osg8Geometry16setTexCoordArrayEjPNS_5ArrayENS1_7BindingE

by the command

> objdump -tT /usr/lib/x86_64-linux-gnu/libosg.so |grep _ZN3osg8Geometry16setTexCoordArrayEjPNS_5ArrayE

which looks similar to the missing symbol above, but the last "NS1_7BindingE" part is additionally there.

How should I resolve this issue? 
Any help, please?

cbfinn, 2017-06-29:
What version of mujoco are you using?

barian-do, 2017-06-29:
it's mujoco131.
Is it mujoco version problem?

pwieler, 2018-08-16:
Hi, I encountered the same issue. Did you find out what the problem was and how to fix it?

pwieler, 2018-08-24:
I could solve the problem in the meantime. The solution that worked for me: I use mujoco131 (since newer versions are not compatible with python2.7) and removed all my existing openscenegraph dependencies. Then i build openscenegraph-3.0.1 from source following the instructions from:
 https://github.com/openscenegraph/OpenSceneGraph/tree/OpenSceneGraph-3.0.1