cbfinn/gps #90: Resource not found: gazebo_worlds

kapsl, 2017-09-18:
Hi,
I finally managed to install gps together with ROS. (I think newer versions then Jade don't work, because there some PR2_Packages missing). And Jade is not running on the newest Ubuntu Versions, so be careful.

I'm Having Ubuntu 14.04 with ROS Jade.

I'm now trying to start the simulated pr2. But I get the error. That ResourceNotFound: gazeob_worlds. I tried, like you explained to edit the pr2_gazebo_no_controller.launch and comment in this line:

<include file="$(find gazebo_ros)/launch/empty_world.launch" />

But still the same error. Any ideas?

Thanks a lot!

kapsl, 2017-09-26:
Ok if ixed that issue by hardcoding the path to empty_world.launch. But i now run into the issue. That the package pr2_gazebo is missing. It seems like it was not released for ROS Jade so it seems like ROS Indigo is needed.

Robotuks, 2017-10-29:
This might help you: [https://github.com/gnoliyil/pr2_kinetic_packages](url)
I am using kinetic and it worked

ruinianxu, 2017-11-06:
Hi Kapsl,
Did you solve the problem of being unable to find gazebo_world? Since the indigo pr2 package is not released, I had to install binary pr2 package separately i.e controller, simulator and so on. However even after installing binary ros-indigo-pr2-simulator, I still met the problem of gazebo world not found.

ruinianxu, 2017-11-07:
> Hi Robotuks,
> I think the issue was solved, by hardcoding the path to empty_world.launch

Hi Kapsl,
Where did you find gazebo_worlds package? Is it in the pr2_simulator
package? Or other package like ros-indigo-gazebo?
Thanks

kapsl, 2017-11-08:
Actually I'm not completely sure anymore. I think it came with the ros gazebo package. I just searched for it in the filesystem and found it...

ruinianxu, 2017-11-08:
Thanks. I solved it. Actually empty_world.launch locates in gazebo_ros
package. And gazebo_ros source code also has some bugs in it. Eventually I
used binary gazebo_ros package.