cbfinn/gps #94: some problem about create more MuJoCo worlds.

facetohard, 2017-10-05:
In the file `agent/agent_mjc.py`, when I tried to create more different worlds in function `_setup_world(self, filename)`, it threw some error like this:

`[[V: not found
sh: 2: [
        : not found
[[V: not found
sh: 2: [
        : not found
[[V: not found
sh: 2: Syntax error: "(" unexpected
[[V: not found
sh: 2: Syntax error: "(" unexpected
ERROR: Invalid activation key`

I am sure that my MuJoCo key is right.
Does any meet this problem?

kapsl, 2017-10-06:
I had the same issue, when i switched to another git branch and tried to run something there?

facetohard, 2017-10-06:
@cbfinn @wmontgomery4 Have you test the code to create different worlds?

cbfinn, 2017-10-08:
Can you be more specific on what you are trying to do?

facetohard, 2017-10-20:
I have created four different files of xml format, so i used the code
`temp = mjcpy.MJCWorld(self._hyperparams['filename'][i])`
line 61 in file ` \path\to\gps\python\gps\agent\mjc\agent_mjc.py`
And then, it threw some errors like what i have posted.