cbfinn/gps #99: 'module' object has no attribute 'RAND_LIMIT'

StrangeTcy, 2017-11-24:
Currently, running `python python/gps/gps_main.py box2d_arm_example`
leads to the following error:
```
DEBUG:No gps_agent_pkg: gps_agent_pkg
ROS path [0]=/opt/ros/lunar/share/ros
ROS path [1]=/opt/ros/lunar/share
Traceback (most recent call last):
  File "python/gps/gps_main.py", line 414, in <module>
    main()
  File "python/gps/gps_main.py", line 348, in main
    hyperparams = imp.load_source('hyperparams', hyperparams_file)
  File "/home/gabriel/gps/experiments/box2d_arm_example/hyperparams.py", line 10, in <module>
    from gps.agent.box2d.arm_world import ArmWorld
  File "python/gps/agent/box2d/arm_world.py", line 2, in <module>
    import Box2D as b2
  File "/usr/local/lib/python2.7/dist-packages/Box2D/__init__.py", line 20, in <module>
    from .Box2D import *
  File "/usr/local/lib/python2.7/dist-packages/Box2D/Box2D.py", line 434, in <module>
    RAND_LIMIT = _Box2D.RAND_LIMIT
AttributeError: 'module' object has no attribute 'RAND_LIMIT'
INFO:signal_shutdown [atexit]
```.
This is probably a problem with my installation of box2d; still, please take a look.

cbfinn, 2017-11-24:
This is most certainly an issue with your box2d installation

StrangeTcy, 2017-11-28:
This is a temporary solution to the issue, and it's not an issue with gps _per se_:

1. Clone this repo
2.  Follow the[ Box2d setup](http://rll.berkeley.edu/gps/), but
do not clone pybox2d (it is seemingly broken at the moment), instead do
`pip install box2d-py`

closed 2017-11-28 by StrangeTcy