cbfinn/gps #107: Unimplemented Geom Type: -1

robintyh1, 2018-07-25:
I am installing gps on ubuntu 16.04. Everything works fine with box2d examples. The compilation for mjcpy is also successful (I have generated mjcpy.so). However, when I execute the example code

``python python/gps/gps_main.py mjc_example``

the error occurs

DEBUG:No ROS enabled: No module named rospkg
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
unimplemented geom type: -1
GLib-GIO-Message: Using the 'memory' GSettings backend.  Your settings will not be saved or shared with other applications.
Segmentation fault (core dumped)

I have not found such error related to mujoco online. Is it a potential error produced by mjcpy? Many thanks for any help.

xiaoziyuan315, 2019-08-19:
Hello, I encountered the same error as you did when I was running the code. I don't know how to solve it. Can you solve it? Looking forward to your reply

xiaoziyuan315, 2019-08-19:
你好，当我运行python python/gps/gps_main.py mjc_example这个命令时，UI界面能出来，但是就是维持几秒，就出现了unimplemented geom type: -1
Segmentation fault (core dumped)错误，然后UI页面就退出了。
请问你解决这个问题了吗？怎么解决的？虚心请教

DEAkimov, 2020-02-11:
Has anyone solved the problem?