cbfinn/gps #113: Why is nominal trajectory not used in iLQR control law?

sklaw, 2019-08-03:
It seems the following line computes new action without any use of nominal trajectory:
https://github.com/cbfinn/gps/blob/82fa6cc930c4392d55d2525f6b792089f1d2ccfe/python/gps/algorithm/traj_opt/traj_opt_lqr_python.py#L257

But in the following papers, nominal trajectory seems to be essential in the whole iLQR algorithm:
[equation (8b) in this paper](https://homes.cs.washington.edu/~todorov/papers/TassaIROS12.pdf)
[equation (3) in this paper](https://graphics.stanford.edu/projects/gpspaper/gps_full.pdf)

dujinyu, 2020-01-07:
@sklaw hello, I am confused by the linear-gaussian policy u = K * x + k, if we use iLQR, why not  the cotrol u = \bar{u} + k + K * (x - \bar{x})

sklaw, 2020-01-07:
> @sklaw hello, I am confused by the linear-gaussian policy u = K * x + k, if we use iLQR, why not the cotrol u = \bar{u} + k + K * (x - \bar{x})

Hi. I'm also confused by this part, too. It seems the author of this code assumed the nominal trajectory is 0... But if so the whole trajectory optimization would be really weird